pires/obd-java-api #30: Issues with DtcNumberObdCommand and TroubleCodesObdCommand

SageKaz, 2015-04-06:
Hello,
While working with your API, I've had great success in general so I have to thank you for that first. Everything seems to work well in general and I haven't had issues with most of the commands.
It's when I get to DtcNumberObdCommand and TroubleCodesObdCommand that issues start to occur.

In the case of DtcNumberObdCommand, after running it it throws the following error:
E/AndroidRuntime(11826): FATAL EXCEPTION: Thread-410
E/AndroidRuntime(11826): Process: com.vuzix.networker, PID: 11826
E/AndroidRuntime(11826): pt.lighthouselabs.obd.exceptions.NonNumericResponseException: Error reading response: 1:06410100040000
E/AndroidRuntime(11826):        at pt.lighthouselabs.obd.commands.ObdCommand.fillBuffer(ObdCommand.java:155)
E/AndroidRuntime(11826):        at pt.lighthouselabs.obd.commands.ObdCommand.readResult(ObdCommand.java:138)
E/AndroidRuntime(11826):        at pt.lighthouselabs.obd.commands.ObdCommand.run(ObdCommand.java:84)

In the case of TroubleCodesObdCommand, the issue I get is the following:
E/AndroidRuntime(12254): FATAL EXCEPTION: Thread-416
E/AndroidRuntime(12254): Process: com.vuzix.networker, PID: 12254
E/AndroidRuntime(12254): java.lang.StringIndexOutOfBoundsException: length=16; index=16
E/AndroidRuntime(12254):        at java.lang.String.indexAndLength(String.java:500)
E/AndroidRuntime(12254):        at java.lang.String.charAt(String.java:494)
E/AndroidRuntime(12254):        at pt.lighthouselabs.obd.commands.control.TroubleCodesObdCommand.performCalculations(TroubleCodesObdCommand.java:82)
E/AndroidRuntime(12254):        at pt.lighthouselabs.obd.commands.ObdCommand.readResult(ObdCommand.java:139)
E/AndroidRuntime(12254):        at pt.lighthouselabs.obd.commands.ObdCommand.run(ObdCommand.java:84)

You may already know of what's going on, but I couldn't say for sure myself since this is really my first step into working with OBDII to any significance.

The last bit of information I can provide is that the vehicle I was testing on was a 2011 Chevrolet Cobalt. This may or may not have some relation to the issue. I am not sure personally. If there's any other way I can help you pinpoint the issues (more information, more testing, ect) just let me know.

Thanks!

pires, 2015-04-06:
`pt.lighthouselabs.obd.commands.control.TroubleCodesObdCommand.performCalculations(TroubleCodesObdCommand.java:82)` throws `java.lang.StringIndexOutOfBoundsException: length=16; index=16`, so that's the issue. Will try to fix tomorrow.

pires, 2015-05-15:
Haven't got the time to fix this. @nomwerp care to give a try?

nomwerp, 2015-05-16:
sorry can't make much sense of it without the data, would it be possible to look into the value of workingData ?        could it possibly be another case of junk data?  in android-obd-reading we are having to strip out some of the junk before performCalculations() is called

https://github.com/pires/android-obd-reader/blob/master/src/main/java/pt/lighthouselabs/obd/reader/activity/TroubleCodesActivity.java

```java
  public class ModifiedTroubleCodesObdCommand extends TroubleCodesObdCommand {
      @Override
      public String getResult() {
          // remove unwanted response from output since this results in erroneous error codes
      return rawData.replace("SEARCHING...", "");
      }
  }

```
we should probable add this to obd-java-api instead so that other projects also can benefit from it

anti43, 2015-06-01:
@nomwerp workingData in case of error is here

performCalculations workingData﹕ "41006000"

java.lang.StringIndexOutOfBoundsException: length=8; regionStart=7; regionLength=3

commit 14a0f630 (2015-07-09): Fixes #30
closed 2015-07-09 by pires