pires/obd-java-api #34: Implement command to read VIN number

jnsmith0274, 2015-06-05:
How about a command to read the VIN number (mode 09)?

pires, 2015-06-05:
Why not? Will gladly review a PR.

anti43, 2015-06-08:
@pires I have it anyway, will do PR tomorrow so you can close this then

pires, 2015-06-08:
@anti43 awesome!

nomwerp, 2015-07-14:
@pires it seems that the response is always "7F0912" for my car , even when all other command return "NO DATA" for example when car not powered, is this a bug?

pires, 2015-07-14:
Probably.

anti43, 2015-07-15:
@pires see https://github.com/pires/obd-java-api/pull/41
it was using just the first buffer chunk before which is always the same

pires, 2015-07-15:
Thank you @anti43! I probably screwed up when cherry-picking your commits before. 

Will release RC3 in a moment.

pires, 2015-07-15:
Tests are failing so I can't release RC3.

```
    Tests run: 42, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 2.154 sec <<< FAILURE!
    testValidSpeedMetricWithMessage(com.github.pires.obd.commands.ExtraMessagesTest)  Time elapsed: 0.011 sec  <<< FAILURE!
    com.github.pires.obd.exceptions.NonNumericResponseException: Error reading response: BUSINIT...410D40
    	at com.github.pires.obd.commands.ObdCommand.fillBuffer(ObdCommand.java:180)
    	at com.github.pires.obd.commands.ObdCommand.readResult(ObdCommand.java:163)
    	at com.github.pires.obd.commands.ExtraMessagesTest.testValidSpeedMetricWithMessage(ExtraMessagesTest.java:78)
```

anti43, 2015-07-15:
@pires IMO the "BUS INIT" message should be entirely ignored (as with the "SEARCHING" message) or throw an according BusInitException. If this test is a valid scenario then the message BUS INIT has not been handled properly in the past. There shall only be two modes of handling 

1. remove message on reading the response
or 
2. throw exception

This test implies that there is a "silently ignore" magic?

pires, 2015-07-15:
@anti43 please run the tests before new PRs. `mvn clean test`. #43 is not working.

anti43, 2015-07-15:
@pires yes probably, as I wrote it was just a quick guess as grounds for discussion.
I did not try this, I just wanted to show you the problem. OK I agree, PR is not a good way to do this, sorry if this was not clear!

anti43, 2015-07-15:
@pires test pass now.
Sorry again:) next time iI use a patch not a PR for untested code
.

https://github.com/pires/obd-java-api/pull/44

closed 2015-07-15 by pires

nomwerp, 2015-07-15:
sorry to be a pain but i'm with RC3 i'm still getting 7F0912..
[image]
..but the unsupported commands now return "7F0112" instead of "7F 01 12"

nomwerp, 2015-07-15:
just checked the raw response is also `7F 09 12 \r\r` so nothing getting lost later

anti43, 2015-07-15:
@pires I suggest to add this

https://github.com/anti43/obd-java-api/blob/master/src/main/java/pt/lighthouselabs/obd/exceptions/UnsupportedCommandException.java

which works for 01 commands. for the 09 commands, this should support 7F0912 as well. so probably a generic UnsupportedCommandException matching a regex for 7F0[0-9]12 should handle this well.

@nomwerp 7F means "not supported" basically.

pires, 2015-07-15:
@anti43 awesome! Thank you.

@nomwerp would you be able to implement a workaround on `android-obd-reader` code to just print `NO DATA` when `7F 09 12` is the command result?

nomwerp, 2015-07-16:
thanks , so not a bug after all ;)

commit c26a2027 (2015-07-09): Implemented VIN command based on anti43 work.
Fixes #34
commit 5771c41a (2015-07-15): Merge pull request #41 from anti43/patch-1

Fixed VIN result parsing. Fixes #34
commit 1a5575ef (2015-07-15): Fixes #34
Fixes #43