pires/obd-java-api #58: OBD ELM327 Device returning NODATA for every request

dnlgby, 2015-08-01:
Before running ant command, i run this code:
        try
        {
            runCommand(sock, new ObdResetCommand());
            runCommand(sock, new EchoOffObdCommand());
            runCommand(sock, new EchoOffObdCommand());
            runCommand(sock, new LineFeedOffObdCommand());
            runCommand(sock, new TimeoutObdCommand(OBD_INIT_TIMEOUT_VAL));
            runCommand(sock, new SelectProtocolObdCommand(ObdProtocols.AUTO));
            runCommand(sock, new AmbientAirTemperatureObdCommand());

        } catch (Exception e)
        {
            throw new ObdConnectionFailure(e.getMessage());
        }

- OBD_INIT_TIMEOUT_VAL = 255

this code ends successfully, no exceptions.

after, i run some commands like:
SpeedObdCommand
EngineRPMObdCommand

And i always get NODATA as a result.

I have no idea what to try.

Thanks in advance.

anti43, 2015-08-02:
your car bus is too slow probably. increase the command delay.

dnlgby, 2015-08-03:
thanks anty, but my timeout val is 255, the max val (one second).

pires, 2015-08-03:
@dnlgby the value is in milliseconds.

dnlgby, 2015-08-03:
when i work with obd terminals i get results, even if i set the timeout to 255.

dnlgby, 2015-08-03:
Its not 255*4, 1020 milliseconds?

anti43, 2015-08-03:
you misunderstood me probably. im not talking about the timeout val, its
the delay inside the ObdCommand class

dnlgby, 2015-08-03:
i see...
But i get response, 'NODATA'. as i see this time its only the time for the request thread to sleep.

dnlgby, 2015-08-07:
changed ResponseTimeDelay to many values. same,

dnlgby, 2015-08-07:
Another applications like torque working well...

pires, 2015-08-31:
I have no idea why this happens.

Pho3niX90, 2015-09-08:
Try removing the speed up part, and test.
https://github.com/pires/obd-java-api/blob/master/src/main/java/com/github/pires/obd/commands/ObdCommand.java#L44

Pho3niX90, 2015-09-09:
Also what vehicle is it? I have noticed specifically diesel bmws have some issues with obd2.

dnlgby, 2015-09-12:
Tried. still the same :(

My vehicle is seat leon fr 2014.

FireWolf2007, 2015-09-14:
what is protocol used?
maximum data on ISO_15765_4_CAN.
Tested on VW Jetta 2013.

closed 2015-09-14 by pires

Pho3niX90, 2015-09-14:
@pires was I correct in saying it was the speed up command?

pires, 2015-09-14:
@Pho3niX90 seems so.

dnlgby, 2015-09-18:
actually not. 

Log:
com.github.pires.obd.exceptions.NoDataException: Error running 01 2F, response: NODATA

Pho3niX90, 2015-09-18:
@dnlgby seems it's only fuel level not working for you? It isn't supported by all cars. Not even toeque has it implemented. But try via a eln327 obd2 terminal

dnlgby, 2015-09-18:
seems like you are right.
working well now.

Thank you!

commit d05346f3 (2015-09-14): Removed bogus support for command results that could be returned by device even before ECU replied.
Fixes #67
Fixes #58
commit 2c5c1194 (2015-11-10): Removed bogus support for command results that could be returned by device even before ECU replied.
Fixes #67
Fixes #58