pires/obd-java-api #61: RC8 release execute error on Windows

greatinfallible, 2015-08-30:
Error:Execution failed for task ':app:preDexDebug'.
> com.android.ide.common.process.ProcessException: org.gradle.process.internal.ExecException: Process 'command 'C:\Program Files\Java\jdk1.8.0_40\bin\java.exe'' finished with non-zero exit value 1

...while parsing com/github/pires/obd/commands/control/DistanceTraveledSinceCodesClearedCommand.class

A think this error is occuring on windows because of the huge-lenght name of this DistanceTraveledSinceCodesClearedCommand.class.
The Windows limits the path length.
Please refactor it's name to a smaller one. e.g: DistTravSinCCCmd.class

Hope it helps!

pires, 2015-08-31:
Yes, this has been identified by other users and I'm very glad to review a PR that fixes it.

greatinfallible, 2015-08-31:
Ok, I will be wainting for the release.
Thank you!

greatinfallible, 2015-08-31:
Pull request made.

commit bf8f6c76 (2015-08-31): Merge pull request #64 from Geraldo-Neto/geraldoneto-patch-1

Fixes #61
closed 2015-08-31 by pires