pires/obd-java-api #66: Parcelable implementation on jobs

tunjid, 2015-09-13:
Fantastic library you've put together, it's much appreciated.

I wanted to write an app based on it, and I looked to the app you have distributed on Github. It appears the service holds a reference to the main activity to execute the jobs in the blocking queue.

I'd rather not do this, as I will have multiple activities / fragments whose UI may need updating. It therefore makes sense to register a Broadcast Receiver in each activity that will need to update the UI when a new job is pushed to it.

To do so, Jobs will need to implement Parcelable so they can be bundled into an intent. I plan to do this sometime tomorrow, and submit a pull request.

What do you think?

Again, thanks for the stellar library.

pires, 2015-09-13:
This library is pure Java, so Android stuff will not be included. Why not provide such implementation with android-obd-reader? Welcome a PR!

closed 2015-09-13 by pires

tunjid, 2015-09-13:
Sounds good, will do that then. Thanks again for the stellar library!

pires, 2015-09-14:
@tunjid thank you for your kind words.