pires/obd-java-api #67: ObdCommand issue

patrickSUP, 2015-09-14:
Hi, 

I noticed that you added a speed up part which consists in adding " 1" after every command, this change causes NODATA response, I tried with 3 cars and it gave the same result. I'm using  OBDLink LX adapter. When I removed it everithing worked fine, can I ask with which setup this speedup works?
Here the part I'm referring to:
public ObdCommand(String command) {
        this.cmd = command;
        this.buffer = new ArrayList<Integer>();
        if (this instanceof ReturnAsapCommand) {
            this.cmd += " 1";//speed up 
        }
    }

pires, 2015-09-14:
@anti43 I'm seeing a lot of users reporting this. Can you chime in and fix it if needed?

anti43, 2015-09-14:
It maybe that not all adapters support it?

pires, 2015-09-14:
@anti43 can we make it optional somehow? Otherwise, I prefer to remove this functionality.

anti43, 2015-09-14:
i think then you must remove it for now, until i found a way to detect the
ability of the adapter

commit d05346f3 (2015-09-14): Removed bogus support for command results that could be returned by device even before ECU replied.
Fixes #67
Fixes #58
commit 2c5c1194 (2015-11-10): Removed bogus support for command results that could be returned by device even before ECU replied.
Fixes #67
Fixes #58
closed 2015-09-14 by pires