pires/obd-java-api #69: Delete DTC code

navat, 2015-09-19:
Hi all, 

I was able to make all functions work on both simulator and real car except clearing the DTC code since I couldn't find the command to do.

You guys have any suggestion on how to delete the DTC code using the api.

Thank you in advance,

sytolk, 2015-09-21:
Hi
I don\`t have success with DTC too but you have use https://github.com/pires/android-obd-reader application or ? And this simulator https://github.com/Hesamedin/ELM327#simulator-mode ?
I\`m trying to get (and clear) car distance traveled (kilometers) do you have success with this command ?

Pho3niX90, 2015-10-02:
This is what you are looking for => https://github.com/pires/obd-java-api/blob/master/src/main/java/com/github/pires/obd/commands/protocol/ResetTroubleCodesCommand.java

I have had success with multiple cars. However, persistent/current codes that haven't been sorted will immediately trigger another error

pires, 2015-12-01:
Thank you @Pho3niX90!

closed 2015-12-01 by pires