pires/obd-java-api #70: Why remove AvailablePIDs command for 21-60?

Dechcaudron, 2015-10-01:
With previous versions of this library, there were 3 commands to check for available commands. Right now all I can see is the **AvailablePidsCommand**, which offers only availability flags for 2-19. Did you remove these commands for any reason?

pires, 2015-10-04:
@Dechcaudron what were those three commands?

Dechcaudron, 2015-10-04:
@pires now that I look at the repo history, it seems they never existed. I probably worked with a modified version of your source. Expect a pull request (if you don't mind me implementing it. It will be my first pull request ever) from me.

In https://en.wikipedia.org/wiki/OBD-II_PIDs#Mode_01 you can see commands 00, 20, 40 and so on cover PID command availability for 01-20, 21-40 and 41-60 respectively. 

Thanks for your attention :+1:

pires, 2015-10-04:
Go for it :+1:

PR #74 (2015-10-06): Available PIDs revamp
Fixes #70 
Fixes #72
closed 2015-10-06 by pires