pires/obd-java-api #71: Fuel trim issue

Pho3niX90, 2015-10-02:
https://github.com/pires/obd-java-api/blob/5dbcf8493720bc6bc831d05ba056e4d72539446f/src/main/java/com/github/pires/obd/enums/FuelTrim.java

These:

    SHORT_TERM_BANK_1(0x06, "Short Term Fuel Trim Bank 1"),
    LONG_TERM_BANK_1(0x07, "Long Term Fuel Trim Bank 1"),
    SHORT_TERM_BANK_2(0x08, "Short Term Fuel Trim Bank 2"),
    LONG_TERM_BANK_2(0x09, "Long Term Fuel Trim Bank 2");

Will yield codes of
6
7
8
9

And will send the command as:
01 6 = for example.
https://github.com/pires/obd-java-api/blob/5dbcf8493720bc6bc831d05ba056e4d72539446f/src/main/java/com/github/pires/obd/enums/FuelTrim.java#L65

I changed that line to 

    return new String("01 0" + value);

pires, 2015-10-02:
Is there a PR?

Pho3niX90, 2015-10-06:
https://github.com/pires/obd-java-api/compare/master...Pho3niX90:patch-1

closed 2015-10-06 by pires