pires/obd-java-api #77: TimeoutCommand(int) is not applicable

410N50, 2015-10-18:
Hi Paulo, I hope that you are fine.
I'm developing an app in Android Studio 1.4 and I'm using your library (obd-java-api). I have an issue when I created "new TimeoutCommand().run(socket.getInputStream(),socket.getOutputStream());"
Error:(134, 21) error: no suitable constructor found for TimeoutCommand()
constructor TimeoutCommand.TimeoutCommand(TimeoutCommand) is not applicable
(actual and formal argument lists differ in length)
constructor TimeoutCommand.TimeoutCommand(int) is not applicable
(actual and formal argument lists differ in length)

Other configuration was created fine Ej.
new LineFeedOffCommand().run(socket.getInputStream(),socket.getOutputStream());
Please, could you help me with this issue?

pires, 2015-10-18:
The example was not as it should be. It's now fixed.

closed 2015-10-18 by pires

410N50, 2015-10-18:
Hi Paulo,
Many thanks for your quick response.
I have other question.
This is my code in Android Studio.
********************************
UUID uuid = UUID.fromString("00001101-0000-1000-8000-00805F9B34FB");
                                                 BluetoothSocket socket = null;
                                                 BluetoothSocket tempsocket = null;

                                                 try {
                                                     socket = device.createInsecureRfcommSocketToServiceRecord(uuid);
                                                     socket.connect();
                                                     Toast.makeText(getApplicationContext(), "Bluetooth conected", Toast.LENGTH_SHORT).show();
                                                     try {
                                                         new EchoOffCommand().run(socket.getInputStream(), socket.getOutputStream());
                                                         new LineFeedOffCommand().run(socket.getInputStream(),socket.getOutputStream());
                                                         new TimeoutCommand(125).run(socket.getInputStream(),socket.getOutputStream());
                                                         new SelectProtocolCommand(ObdProtocols.AUTO).run(socket.getInputStream(),socket.getOutputStream());
                                                         new AmbientAirTemperatureCommand().run(socket.getInputStream(),socket.getOutputStream());
                                                         Toast.makeText(getApplicationContext(),"Finalizó opcion try luego de comando OBD" ,Toast.LENGTH_SHORT).show();

                                                     }catch (Exception ex){
                                                         Toast.makeText(getApplicationContext(),"Message after OBD command" ,Toast.LENGTH_SHORT).show();

                                                     }

                                                 } catch (Exception e) {
                                                     // TODO Auto-generated catch block
                                                     e.printStackTrace();
                                                     Toast.makeText(getApplicationContext(), "Bluetooth error", Toast.LENGTH_SHORT).show();

                                                 }
**************************************
I tested in a car and socket connection is fine because I received the messages "Bluetooth conected" and "Message after OBD commands".
but i dont know how to capture for Ex. AmbientAirTemperatureCommand() in order to do math operation,
Any suggestion?
Thanks in advanced.

commit 8d61fd4c (2015-10-18): Fixed TimeoutCommand instantiation example. Fixes #77