pires/obd-java-api #95: responseTimeDelay can be set but is ignored

stealthcopter, 2016-02-23:
In ObdCommand, there is a method setResponseTimeDelay(long delay) to set the responseTimeDelay, however this value assigned but is never used. I think there should probably be a wait after sendCommand(..) to ensure we are not trying to read a response too soon as sometimes it's necessary to set the delay quiet high when working with older protocols.

pires, 2016-02-23:
@stealthcopter care to provide a PR that fixes it? Glad to review. Don't forget the tests. Thank you.

stealthcopter, 2016-02-23:
#96 I've made a pull request with the changes, I'm not familiar with real java development / test, so I've done my best to show how I'd expect the test to look.

stealthcopter, 2016-02-25:
Fixed as of #96

closed 2016-02-25 by stealthcopter