pires/obd-java-api #97: Obd2 fault codes

xavierDesarrollo, 2016-02-25:
Hi, OBD2 sends you fault codes, how I do to see these codes on android eclipse or what kind of class or method of obd_java_api can i to use???? please help me.

I need to connect to obd2 by bluetooth and get wathever fault code that obd2 throw me and show in my android.

pranaypatel512, 2016-04-11:
Can you tried any code using obd_java_api?

xavierDesarrollo, 2016-04-11:
hi pranaypatel, no i didn't.... please help me, how can i use thouse codes??

Pho3niX90, 2016-05-11:
@xavierDesarrollo please have a look at https://github.com/pires/android-obd-reader for an android example. 
The classes you would use are 
https://github.com/pires/obd-java-api/blob/master/src/main/java/com/github/pires/obd/commands/control/TroubleCodesCommand.java
https://github.com/pires/obd-java-api/blob/master/src/main/java/com/github/pires/obd/commands/control/PendingTroubleCodesCommand.java
https://github.com/pires/obd-java-api/blob/master/src/main/java/com/github/pires/obd/commands/control/PermanentTroubleCodesCommand.java

pires, 2016-05-11:
Thank you @Pho3niX90.

closed 2016-05-11 by pires