pires/obd-java-api #98: SpeedCommand always returning speed of 52 even when not moving

Abushawish, 2016-03-17:
Using ELM327. I think it may have something to do with the hex code that's being returned/parsed incorrectly. Thoughts?

pires, 2016-03-17:
Probably some protocol issue.

frandrago, 2016-03-29:
Hi, I do not know how to write in Inglés, but I am going to try.

I have the same problem, you've found the solution?

Thank you!!

frandrago, 2016-04-04:
I have this problem when I use the Bluetooth ELM327 V2.1 ODBII.
With earlier versions do not have this problem.

Pho3niX90, 2016-05-11:
@Abushawish @frandrago 
Could you please supply your rawData value? 

Easiest to do the above is edit the SpeedCommand.java
`public String getCalculatedResult() {
        //return useImperialUnits ? String.valueOf(getImperialUnit()) : String.valueOf(getMetricSpeed());
        return rawData
    }`

phamels, 2016-05-15:
@Pho3niX90 

Had the same issue using android-obd-reader (which uses obd-java-api)
The speed was displayed as a constant 65km/h while standing still and engine RPM of 4163.
Also the runtime was a constant 04:37:51.

A similar application (not to use any names) uses the protocol ISO15765-4CAN11/500 on this car & adapter combination but returns the data just fine.

Included the obd-java-api into the android-obd-reader project and changed the getCalculatedResult() to return the rawData though that didn't change anything to the data being displayed on the android-obd-reader app.

Anything I can do to test / help / give more data / results?

Pho3niX90, 2016-05-16:
@phamels the library does not support the can protocol properly. So that would be your reason, however hopefully in the next month it will ;)

pires, 2016-05-16:
@pho3nix90 that would be truly amazing.

phamels, 2016-05-16:
@Pho3niX90 aha ok, expected it would be something like that :-) that would be awesome :-) This is my first dive into the OBD protocols but if there's anything I could do to provide feedback or help in any way, let me know!

Pho3niX90, 2016-05-16:
Thanks for the offer, it is taking longer than I initially anticipated, since all my vehicles are K-LINE, and cannot find a simulator that communicates solely in CAN. Thus making testing difficult. I will post a pull request for the speed command first to make sure it works for everyone.

Pho3niX90, 2016-05-16:
@phamels if you could do me a favour though, if you could install a obd2 terminal app on your phone, and send a command to the OBD2 adapter, could you please post the response from the vehicle up?+

phamels, 2016-05-16:
@Pho3niX90 was actually already preparing my laptop as you wrote this to do this :-) heading down to the garage as we speak to to that for you :-)

phamels, 2016-05-16:
@Pho3niX90 as promised:

TeraTerm: 

`
>ATI
ELM327 v2.1
>ATZ
ELM327 v2.1
>ATSP6
OK
>ATMA
STOPPED
>ATCS
T:00 R:00 
>ATRV
11.2V
>STI
?
>0100
41 00 BE 3E A8 13 
>011c
41 1C 06`

Android OBD Terminal:

`
> ATZ
ATZ

ELM327 v2.1

> ATH1
ATH1
OK

> ATDP
ATDP
ISO 15765-4 (CAN 11/500)

> ATMA
ATMA
STOPPED

> 010D
010D
7E8 03 41 0D 00 

> 010C
010C
7E8 04 41 0C 00 00 

> 010C
010C
7E8 04 41 0C 0A A0 

> 010C
010C
7E8 04 41 0C 0A 88 

> 010C
010C
7E8 04 41 0C 0A B4 
`

ATMA doesn't seem to return much data, neither on the android terminal or on the TeraTerm using windows

Pho3niX90, 2016-05-16:
Was your RPM around 685?

phamels, 2016-05-16:
Pretty sure it should be around there as I was just idling

Pho3niX90, 2016-05-16:
OKay. Could you please try that again for me. 

First turn off the headers
ATH0

Then do RPM
010C

phamels, 2016-05-16:
I'll go down in a minute to do that and take note of the RPM on the dash while doing it :-)

Pho3niX90, 2016-05-16:
Thank you! Also, which vehicle are you testing this in, and which country?

phamels, 2016-05-16:
My pleasure! Currently doing it on a 2010 Skoda Yeti (Imported it from Belgium; now doing it/living in Sofia, Bulgaria) (also have a BMW 640d available as well for dumping if required)

Pho3niX90, 2016-05-16:
Awesome thanks. The API should work flawlessly with the 640d, unless it's a 2008+ model, then can might become an issue. However, as far as I know the ECU stayed the same for the E63 up until 2010

phamels, 2016-05-16:
This one is a F13 :-) could do some tests on that as well :-) Heading down now to do a dump

Pho3niX90, 2016-05-16:
Thanks, yeah F13 might be a problem. Would be great if you can do tests on it. And we can fix both in one go. The N57D30 is a brilliant engine.

phamels, 2016-05-16:
In the car right now :-)

First engine off; then on at idle and at the last few tried to keep it around 1000rpm give or take a little:

>> 010C
010C
41 0C 00 00 

>> 010c
010c
41 0C 0A B8 

>> 010c
010c
41 0C 0A 78 

>> 010c
010c
41 0C 0F BC 

>> 010c
010c
41 0C 10 54 

>> 010c
010c
41 0C 10 A4 

>> 010c
010c
41 0C 10 8C 

I'll check for the F13 as well asap (not in the garage right now :-))

Pho3niX90, 2016-05-16:
Hmm strange, from those results the library should work perfectly.
Have you tried the latest build of the app?

phamels, 2016-05-16:
Seems to be correct; as far as I've been studying the output: 
PID 0C which returns RPM in this case 10 8C converted to decimal 4236 divided by four returns 1059 RPM; correct?

phamels, 2016-05-16:
Yeah I have, compiled the source yesterday

Pho3niX90, 2016-05-16:
The equation is as follows:
So output is: 
010c
41 0C 10 8C

410C gets removed, since it's just an echo of the command sent, you are left with 2 bytes, 10 and 8C, converted
10 = 16
8C = 140

Full equation is = (A * 256 + B) / 4  = 1059.

Pho3niX90, 2016-05-16:
OKay could you set the protocol in the app to KWP, as a test

Pho3niX90, 2016-05-16:
I think what is happening is the following. The terminals are infact not setting to automatic but hardcoding the protocol to some kline protocol, thus the vehicle is responding in the same manner. Where the app is by default on AUTO and the car is expecting CAN messages. This however is just a theory

phamels, 2016-05-16:
Aha, thanks for the clarification on that! Just started my exploration into all of this since yesterday :-)

I'll do that, head down and come back with the results!
I tried yesterday using AUTO and ISO_15765_4_CAN with same results.
I'll try the ISO_14230_4_KWP now and perhaps a few others as well as a matter of testing

Pho3niX90, 2016-05-16:
Yes that would be great. Any of the protocols that do not contain CAN. 
I will still however try and get the CAN working, since it is a way faster protocol.

phamels, 2016-05-16:
Damn, KWP displays all data correctly! How could I have missed trying that! I will try to do some testing on the F13 though asap

phamels, 2016-05-16:
Oh, can try with the F13 now if you like? What could I dump for you?

phamels, 2016-05-16:
Weird, tried again now and getting businit error using kwp and the errorous values with can

Pho3niX90, 2016-05-16:
Hmm strange. I think the app is sending the commands too fast. Plug out the adapter and back in, en check if you can modify the sleep time in the settings to something a little but higher, you could try one of the other non CAN protocols as well. Yes if you can send the same test scenarios for the F13 that would be great.

Pho3niX90, 2016-05-16:
When you received the businit error, was it via the android compiled app or your own app?

phamels, 2016-05-16:
The Businit error was with the android compiled app which was working earlier.
Tried different protocols again, ATZ command then trying it, restarting the phone even just to be sure :-) but now kept receiving protocol error where it was working before, strange indeed :-)
Also tried the F13; actually just gives the exact same values and uses the exact same protocol as the Skoda Yeti :-)

Pho3niX90, 2016-05-16:
That is strange. And if you go back to the terminal after the app give you a buss error, does it still bring back results?

phamels, 2016-05-16:
it is strange indeed, yeah it actually does when I try to use the terminal after the app.
Going to look more into it this evening and see what's going on there

Pho3niX90, 2016-05-16:
Okay great. Will have a look on my side as well what would cause the error in the interim, until I can get a working version of the CAN protocol.

phamels, 2016-05-16:
Can't get it to work on KWP at all anymore. Not in the app, not on terminal (also get the bus errors there using atsp4 and trying to fetch data)

phamels, 2016-05-16:
Didn't test it before though with atsp4 on terminal, only atsp6. Kind of weird however i got the app to display correct data before with kwp including VIN number and all

Pho3niX90, 2016-05-16:
I am going to implement a tester in my app to try all protocols from 1 to 7 to automatically find the correct protocol, instead of leaving it to the hardware. I eill uploaf the apk here for you to test if that is okay? If all is well I will do a PR.

phamels, 2016-05-16:
sounds good! pretty much went through all the protocols in the terminal; the one that worked for me was number 6 (using the AT command to select the protocol 6 (ATSP6) or the auto one (ATSP0)) the rest threw up bus errors.

Pho3niX90, 2016-05-16:
and on protocol 6, does it return in the same format you posted results earlier?

phamels, 2016-05-16:
Yeah, that's what it did all along; it only gave me that output on the CAN protocol.
I'm still puzzled why it even worked on the KWP in the app :-)

Pho3niX90, 2016-05-17:
Hmm weird. And when you set the app to protocol 6, and then try to read data what does it do?

Pho3niX90, 2016-05-17:
@phamels 
Could you try this one on the vehicle? It uses the same library that has been modified to pass Parcels. 
However, it's a highly modified version of the original android project, and have also included to display the raw responses from the vehicle in the logcat, that will help diagnose why you are getting the errors. 

[Hazmat-Reader-debug-0.8.0.zip](https://github.com/pires/obd-java-api/files/267866/Hazmat-Reader-debug-0.8.0.zip)

phamels, 2016-05-17:
When I set it to protocol 6 (or ISO_15765_4_CAN) in the app; I get the wrong values; as in a fixed speed of 65km/h at 4163RPM with a fixed trip time of 04:37:51 and a vehicle VIN with funky chars :-)

I'll go ahead and tre APK you attached!

Pho3niX90, 2016-05-17:
Okay thank you, that would explain what is happening if you can extract the logcat.

phamels, 2016-05-17:
Logs have been mailed!

Pho3niX90, 2016-05-17:
Thank you! I think we might have caught the error. 

Are all 4 logs  from the same vehicle? 
It seems like the adapter is returning an echo of the command being sent. I am going to add a fix now

phamels, 2016-05-17:
No, thank you! 
yes they are all from the same vehicle :-) I just tried different settings and one was trying to do error code check but that just crashed the app completely :-)

Pho3niX90, 2016-05-17:
Examples that the vehicle/adapter returned:
Long Term Fuel Trim Bank 1: Calc = -49.2% | Raw = **0107 41 07 80**
So what is happening above is that it's echoing back the command, and adding the header. 
0107 is the command that was sent
4107 is the command that was sent
80 is the actual value.

So the value should have been 0%, meaning the car had to make no adjustments to the fueling, which is correct for a car idling. 

BUT. The app assumes the headers are turned of because we told the adapter to do so, however, it did not. Thus the app is using the 41 as the value, where it should have used 80 as the value. 

I hope that makes sense? 

Here  are more examples:
Engine Runtime: Calc = 04:37:51 | Raw = **011F 411F 0000**
Engine runtime has never worked for me.
 
Throttle Position: Calc = 25.5% | Raw = **0111 4111 2B**
Should have been 16.8, most manufacturers report around 16% for throttle on idle, and 85% at WOT.

Pho3niX90, 2016-05-17:
Here is the weird part, the adapter reported that it did infact turn of the headers, however, it did not , see below:
D/CMDTEST (23691): Headers disabled: Calc = **ATH0OK** | Raw = ATH0OK **<- Reporting back header of ATH, should have been OK only** 
D/CMDTEST (23691): Adaptive timing set: Calc = **ATAT1OK** | Raw = ATAT1OK **Still sending header of ATAT1, should have been OK only**

phamels, 2016-05-17:
aha yes, that does make sense :-) hmm, I'll have to look into the adapter in that case, because at one point when it was working, it actually did turn off the headers as it reported correct data then...
Though I did try to reset it manually etc, I can do some terminal tests to see if I can make it stop echoing

phamels, 2016-05-17:
Well, that or modify the library and see if it is replying with the headers and then just strip them

Pho3niX90, 2016-05-17:
Yes I am adding a check to strip data, to make sure it accounts for the different adapters. 

Give me about 30mins and will have a working test.

phamels, 2016-05-17:
Was able to actually turn off the echo and stop the headers with the terminal; though did notice something when running the app.Where it expects the output for: Reset OBD: it replies with: +CONNECTING
Where it expects the Echo Off: it returns: ATZELM327v2.1
Then where it says: Line Feed Off: it receives ATL0OK

So I'm suspecting that it's not waiting long enough or just seeing the +CONNECTING and ATZELM327v2.1 or reply from the adapters first connection as OK's for the issued commands to turn off the echo and headers and those commands actually are not processed as it is still connecting...

Could that be?

phamels, 2016-05-17:
I think it's exactly that as where in the "Echo Off:" It returns runs ATZ and the adapter replies with ELM327v2.1 where it should have been sending ATE0 and expecting "OK" as a reply, no?

Pho3niX90, 2016-05-17:
That wad my initial though as well. But you will see from the logs you sent me that it did reply OK to the headers off comand, however the command thereafter it still sent the headers.

[12 more replies]

closed 2016-05-18 by pires