pires/obd-java-api #102: Regarding FuelEconomyObdCommand

PareshMayani, 2016-04-05:
Hi,

First of all thank so much for writing OBD Api and Reader. You are awesome!

Now, I have got a project having FuelEconomyObdCommand and using old OBD Api. I have checked commit history and found out that you have removed this class. https://github.com/pires/obd-java-api/commit/558eb16ff9627e9012b70732f03078847c9ebd68

Could you please share more details it? Any replacement class available for the same?

Thanks,
Paresh Mayani

pires, 2016-04-05:
Fuel consumption rate is never implemented by manufacturers. Having that class meant that I tried to somehow get valid data, but I never couldn't. I have removed everything related to fuel consumption.

closed 2016-04-05 by pires

PareshMayani, 2016-04-05:
Thank you @pires for the quick reply. Really appreciate it!