pires/obd-java-api #103: Commands/APIs for detecting level of different types of oil

pranaypatel512, 2016-04-11:
Hi Pires,

Thank you for creating this awesome API, it's really helpful.

I am wondering if there is any command or code through which we can detect level of different oils in car like engine oil, brake oil and gear oil. 

Yes we have gone through the current commands available and have found a command about oil temperature [OilTempCommand.java](https://github.com/pires/obd-java-api/blob/master/src/main/java/com/github/pires/obd/commands/engine/OilTempCommand.java).

And we have found out a fault code [P0521](http://www.obd-codes.com/p0521) from [http://www.obd-codes.com](http://www.obd-codes.com) for _Engine Oil Pressure Sensor/Switch Range/Performance_

Could you please share more details, command or APIs through which we can detect oil level? Is it possible? 

Really appreciate your time and effort. 

Thanks,
Pranay

pires, 2016-04-12:
According to [OBD-II PID list](https://en.wikipedia.org/wiki/OBD-II_PIDs) only `Oil Temperature` is available.

closed 2016-04-12 by pires

pranaypatel512, 2016-04-12:
Thank you @pires for the information.It will be very helpful for me.