pires/obd-java-api #104: Vehicle Identification Number (VIN)

pranaypatel512, 2016-04-14:
is there any command or code through which we can get [Vehicle Identification Number (VIN)](https://en.wikipedia.org/wiki/OBD-II_PIDs#Mode_09) ?

Thanks,
Pranay

Ewg777, 2016-04-14:
@pranaypatel512 try VinCommand but VIN may be not available for some reason.
https://github.com/pires/obd-java-api/blob/master/src/main/java/com/github/pires/obd/commands/control/VinCommand.java

pires, 2016-04-14:
Thank you @Ewg777.

closed 2016-04-14 by pires

pranaypatel512, 2016-04-14:
@Ewg777  Thanks for suggesting command.
I got  Vehicle Identification Number using below code:
`if (cmdID.endsWith(AvailableCommandNames.VIN.toString())) { //cmdID = obdCommandJob.getCommand().getName();
                VinCommand command = (VinCommand) job.getCommand();
Log.e("COMMAND:", "VIN No:" + command.getFormattedResult());
}`

**O/P**
https://goo.gl/6Hv8lt