pires/obd-java-api #106: How to re-establish obd2 connection?

410N50, 2016-04-28:
Hi Paulo, I hope that you are fine.
I'm using your library in order to send OBD commands and receive the response. This process is working fine for me.
In my app, I need to check OBD responses always.
Below my question: 
How to re-establish obd2 connection (automatically) if vehicle turn off and then turn on?
Any suggestion?
Thanks in advanced.

pires, 2016-04-28:
On your app code, you need to detect if the I/O streams are closed and reopen them. I'm not sure the ELM device is turned-off, though so this is just one idea.

closed 2016-04-28 by pires

410N50, 2016-05-17:
Hi Paulo,

ELM is not turn off though car is off.
How to detect if the I/O streams are closed and reopen them?
When car is off I am receiving NODATA in your fillerBuffer function.
Many thanks for your comments.