pires/obd-java-api #117: Erroneous Air Temperature Measurements

KirbyJames, 2016-06-13:
I've been successfully using the OBD-II software and the OBD-Java-Api in 10+ vehicles in the UK.
I've just tried these Apps in two Hertz rental cars in the States
- a 2016 Nissan Versa, and
- a 2016 Hyundai Santa Fe
in both cases they report erroneous Air Temeratures - athough the Air Pressure appeared realistic.

The Nissan reported a constant temperture of +8C over a three hour period although the dashboard was showing temperatures in the range 20-25C.

The App in the Hyundai showed temperatures mostly 10C lower than the actual Air Temperature - although at one point it moved rapidly up to 20C - which was still 4-5C too low.

Are there any reasons why rental cars might not report correct values?
Alternatively are manuafacturers 'obfuscating' the output from OBD ports on newer vehicles as people try to monetize their output?

Pho3niX90, 2016-06-14:
I have geard before that rental companies disable OBD on vehicles. However I do not think this is the case. 

Could you check what the raw output is from the vehicles to better understand what is happening?

pires, 2016-07-16:
Closing due to the lack of feedback.

closed 2016-07-16 by pires