pires/obd-java-api #118: Invalid Trouble Codes

jcrjaci, 2016-09-29:
Hi,
Firstly, great library it help me and simplify a lot.

I have an issue, when  i try to access to the trouble codes of the car by:
```
TroubleCodesCommand dtc = new TroubleCodesCommand();
dtc.run(mSocket.getInputStream(), mSocket.getOutputStream());
promise.resolve(dtc.getFormattedResult());
```

it always return me the following 3 codes:
```
U3EAR
U0HIN
U3...
``` 

After the 'U', it should return an hexadecimal, am i right?
like U3000

Obrigado!

Tommixoft, 2016-09-29:
Looks like creator doesn't know answer

pires, 2016-09-30:
You are right. This may be something related to CAN or your device. 

@tommixoft this is an open-source, non-profit effort of mine. I try to help when I can on my free time. I suggest you do the same instead of wasting your free time trolling.

jcrjaci, 2016-09-30:
@pires I think i found the problem.

I put the starting commands in a function. (EchoOffCommand,LineFeedOffCommand, TimeoutCommand, SelectProtocolCommand).

And i was calling that function and the DTC function synchronous.

now i'm calling both asynchronous. 
and i get a empty string as a answer.
I also called DtcNumberCommand which return 0.
So i think that problem are solved.

closed 2016-09-30 by jcrjaci

pires, 2016-09-30:
Awesome! Thanks for the update.