pires/obd-java-api #119: Protocol Close command

harmittaa, 2016-09-30:
Thanks for the API.

It might be a good idea to add the AT PC command.

> If a connection is lost, you will need to tell the ELM327 to ‘close’ the current connection, with a Protocol Close command (AT PC). This will ensure that the ELM327 starts from the beginning when the next request is made. This is particularly important for the ISO 9141 and ISO 14230 protocols, as they need to send a special initiation sequence.
> 
> Once the protocol has been closed, it can be re-opened by making a request such as 01 00 (do not send ATZ or AT SP0, as many do).
Source: https://www.elmelectronics.com/help/obd/tips/#327_Commands

pires, 2016-09-30:
Thank you @harmittaa, valuable tip indeed!

pires, 2016-09-30:
Man, that link is pure gold!

commit 9b690a27 (2016-09-30): Added protocol close command. Fixes #119
closed 2016-09-30 by pires