pires/obd-java-api #121: First Connect works, the second throws an error

amihaylo, 2016-11-22:
Hi Pires, thanks for providing the obd api it's helped me tons!

I hope this is right way of going about asking for help, if not please redirect me and I will do so. I have had the issue for a while and am out of ideas which is why I turned to you in hopes of figuring out a solution to my problem.

So the first time I connect to OBD I have no issues and read the data just fine. If I however disconnect and try to reconnect, it fails and throws the following error:
`W/System.err: com.github.pires.obd.exceptions.UnableToConnectException: Error running 01 46, response: ...UNABLETOCONNECT`

After that failed attempt, if I try to reconnect again it works.

After every connection I make sure to close it with the **Close Command** as such:

    CloseCommand closeCommand = new CloseCommand();
    closeCommand.run(this.mmSocket.getInputStream(), this.mmSocket.getOutputStream());

Is there something I am not doing correctly? Possibly closing the streams? If so how would I go about doing that?

pires, 2016-11-23:
It may be something with your device.

amihaylo, 2016-11-23:
I've tried it with a number of devices, same error persists. Any ideas what else the issue may be?

pires, 2016-11-23:
`01 46` is Ambient Air Temperature. what happened to all the AT commands needed to reconnect?

amihaylo, 2016-11-23:
Perhaps I'm not doing it properly, but upon every reconnect I initialize with the following commands:

    new EchoOffCommand().run(this.mmSocket.getInputStream(), this.mmSocket.getOutputStream());
    new LineFeedOffCommand().run(this.mmSocket.getInputStream(), this.mmSocket.getOutputStream());
    new TimeoutCommand(125).run(this.mmSocket.getInputStream(), this.mmSocket.getOutputStream()); 
    new SelectProtocolCommand(ObdProtocols.AUTO).run(this.mmSocket.getInputStream(), this.mmSocket.getOutputStream());