pires/obd-java-api #129: How use library with serial port

AndroFlo, 2017-03-08:
Hello, 

How can i use the library with a serial port  like /dev/ttyS0 instead of use a socket ?

Thank's

pires, 2017-03-08:
This library doesn't care about sockets and only cares about `InputStream` and `OutputStream`.

closed 2017-03-08 by pires

blarosen95, 2017-04-12:
@pires "Doesn't care" as in "is incompatible" or is there a way to create the Streams from an existing serial port connection?
I'm struggling to figure out how to get your library properly connected to an ELM327 (or any similar) device. I don't want to open a new "issue" as this isn't an issue on your end per say, more of a lack of knowledge on my half/documentation on yours. I've tried going through the test files to see if I could figure it out, but no such luck. Would really appreciate an example 😄!

Osslack, 2017-04-13:
@blarosen95 

The "getting connected"-part is not part of this library but rather "ordinary" android stuff.
Maybe this can help you:[https://developer.android.com/guide/topics/connectivity/bluetooth.html](url)

You need to connect as a client. mmSocket in the ConnectThread is what you can use to communicate with the device.

I used the UUID mentioned in this guide [http://blog.lemberg.co.uk/how-guide-obdii-reader-app-development](url).

Once you have the socket you simply need to run the initializing commands mentioned in the README of this repository.

`try {
  new EchoOffCommand().run(socket.getInputStream(), socket.getOutputStream());
  new LineFeedOffCommand().run(socket.getInputStream(), socket.getOutputStream());
  new TimeoutCommand(125).run(socket.getInputStream(), socket.getOutputStream());
  new SelectProtocolCommand(ObdProtocols.AUTO).run(socket.getInputStream(), socket.getOutputStream());
  new AmbientAirTemperatureCommand().run(socket.getInputStream(), socket.getOutputStream());
} catch (Exception e) {
  // handle errors
}`

I hope this helps you getting started, this was the stuff I struggled with in the beginning.

pires, 2017-04-13:
The library doesn't care about the transport protocol you're using. It doesn't care if you have a Java app or Android app, if you are connected through serial, wifi, bluetooth. It just cares that in your app an I/O stream pair is accessible so it can read/write from/to the connection.