pires/obd-java-api #130: obd-java-api with SocketCAN?

karu2003, 2017-03-21:
Good afternoon,

I'm developing the CAN-Bus controller for Android using SocketCAN.
Does work library obd-java-api with SocketCAN?

Thank you.

Andrew.

Odmink0, 2017-03-21:
95% - Yes. If your controller like ELM327 and understands it's syntaxis.

karu2003, 2017-03-21:
I do not use ELM327. I use RAW CAN (candump & co).

Odmink0, 2017-03-21:
obd-java-api used for connect to car's CANbus via diagnose scanner based on ELM327 chip and firmware.

karu2003, 2017-03-21:
Clear. Tell me please. Is it difficult to change to SocketCAN? SocketCAN dont need AT comand.

Odmink0, 2017-03-21:
I think that it is impossible. obd-java-api developed especially for using with AT comands.

karu2003, 2017-03-21:
If your add SocketCAN, can be used with RPi and more.  :)

Odmink0, 2017-03-21:
Read name of project: obd-java-api.
obd == OBD == On-board diagnostics => https://en.wikipedia.org/wiki/On-board_diagnostics

karu2003, 2017-03-21:
ELM327 comand are very similar to the RAW CAN frame. I think it will not be difficult.

Odmink0, 2017-03-21:
So you can fork that project and add any options and any frame format.

pires, 2017-03-22:
In theory the library can do whatever you want. Feel free to open one or more pull-requests that add features and related tests and I'll take a look :)