pires/obd-java-api #132: Command to read odometer/mileage reading

vijquick, 2017-04-19:
There should be a command to read the odometer current reading. Have browsed through the source code and couldn't find any command that can return the odometer reading.

Odmink0, 2017-04-19:
OBDII protocols doesn't have milleage PID.

alamirah93, 2017-04-21:
I working on project that should get the odometer/mileage reading!
how can I receive that ?

Odmink0, 2017-04-21:
It is IMPOSSIBLE.
Please read about OBD PIDs(https://en.wikipedia.org/wiki/OBD-II_PIDs)

Maybe some auto vendors include odometer data in own protocol(or bus) but for reading that data you can't use ELM327 adapter.

obd-java-api only for ELM327 adapters.

pires, 2017-04-21:
Thanks for explaining!

closed 2017-04-21 by pires

alamirah93, 2017-04-21:
OK thank you for your answer, do you know another adapter that can read the odometer?

Odmink0, 2017-04-21:
Main problem not in adapter.
In 95% autos milleage data stored in EEPROM chip which not connected to diagnostic bus.
This is made for security purposes.