pires/obd-java-api #136: Output of data

Dlavrov83, 2017-08-24:
Hi,
While searching for a good OBD communicator / connector I came across your project.
It seems pretty straight forward and simple to use (haven't gotten to deep into it yet).
However, i have one small question. While the `ObdCommand.readResult()` is invoked, where do the decoded output goes to? 
Is it a log file, or is it being printed on to the screen?

Thank you.

Odmink0, 2017-08-24:
Hi.
You can call "readResult()" at variable init: 
Sample:

`try {
            ObdRawCommand custom = new ObdRawCommand(command); //command can be any ELM command like "atz" or "0130"
            custom.run(inputs, outputs); //inputs - input stream, outputs - output stream
            String result = custom.getResult(); // your variable with result
        }
catch(Exception e){

}`

Dlavrov83, 2017-08-24:
Ok, thanx.

Same will works if i will try to use of of the other command classes? 
example:
`SpeedCommand spd = new SpeedCommand();
spd.run(In, Out);
String result = spd.getResult();`

By the way, in the example you gave me the command should be a string with out any spaces?

Odmink0, 2017-08-24:
1) Your examples correct.
2) ELM327 can understand commands with spaces and without spaces. "01 00" equals "0100".

Dlavrov83, 2017-08-24:
Thank you very much.

Odmink0, 2017-08-24:
You are welcome.
If you got another questions - you can asking =)